dbisu/pico-ducky #155: Web Interace Encodes Special Characters that are in the Payloads

responsivereality, 2023-03-18:
When I upload a payload onto the Raspberry Pi Pico W with the web interface such as this:

DELAY 2000
COMMAND SPACE
DELAY 1000
STRING https://www.youtube.com/watch?v=dQw4w9WgXcQ&ab_channel=RickAstley
DELAY 1000
ENTER

It turns into this because of the special characters like slashes and question marks and colons that are getting encoded:

DELAY 2000
COMMAND SPACE
DELAY 1000
STRING https%3A%2F%2Fwww.youtube.com%2Fwatch%3Fv%3DdQw4w9WgXcQ%26ab_channel%3DRickAstley
DELAY 1000
ENTER

Is there any fix? IE a way to stop the encoding when payloads are send over internet, because this is definitely not an issue when directly uploading a payload when in setup mode. And this is a rickroll, I know!

dbisu, 2023-03-18:
I think I know what is going on, might take a bit to figure out the best way to fix it.
Are you seeing this only when you upload, only when you use the live edit, or both?

responsivereality, 2023-03-18:
I press “new payload” then write it, and hit create payload, all is fine, until I press edit payload, when I see that all of the special characters are encoded. When I press run payload, all of the special characters are also encoded

dbisu, 2023-03-19:
So the problem is the POST data is getting URL encoded.
You could try changing webapp.py line 34 from:
`<form action="/write/{}" method="POST">`
to
`<form action="/write/{}" method="POST" enctype="text/plain">`
and line 51 from:
`<form action="/new" method="POST">`
to
`<form action="/new" method="POST" enctype="text/plain">`

My Pico W isn't currently setup to use for testing.

responsivereality, 2023-03-19:
Thanks for the quick reply. I just tried that, and after rebooting the device, it was working until I hit the submit button after writing the payload. It was stuck loading for ages, then when I reverted webapp.py back to normal, it never got stuck. Long story short, it did not work.

dbisu, 2023-03-19:
OK, thanks for trying.
It will probably be a bit until I have a fix.

responsivereality, 2023-03-19:
No problem. Thanks for trying as well.

FRLoneWolfFR, 2023-03-19:
@responsivereality I made a fix for that here :
https://github.com/dbisu/pico-ducky/pull/150

responsivereality, 2023-03-19:
I don't understand, would you mind telling me how to fix it? Thanks.

FRLoneWolfFR, 2023-03-19:
Replace webapp.py by this https://github.com/dbisu/pico-ducky/blob/09bf09033f2b771fbea45b0da0bd3fe80ed3c474/webapp.py

responsivereality, 2023-03-20:
Thank you so much! It now works perfectly. @dbisu, would you mind if I upload an updated tutorial video on youtube so that more people can use this brilliant tool of yours? Of course, I would credit you entirely.

dbisu, 2023-03-21:
Sure, no problem on the video.

It will probably be a bit on merging this change in. Working on a different project this week. hopefully by the weekend I can verify the fix and get it merged.

closed 2023-03-26 by dbisu