dbisu/pico-ducky #160: .uf2 Not Working

NonTrusted, 2023-03-30:
When i copy .uf2 file to the Pi Pico, It does reconnect as "CIRCUITPY" but it doesn't have the files i need, it shows "boot_out.txt" only and there is:

Adafruit CircuitPython 8.0.4 on 2023-03-15; Raspberry Pi Pico with rp2040
Board ID:raspberry_pi_pico

Can anyone help????

dbisu, 2023-03-31:
This has been seen before. I think it must be an issue in the CircuitPython loader.
Follow the instructions in [RESET.md](https://github.com/dbisu/pico-ducky/blob/main/RESET.md)

NonTrusted, 2023-03-31:
> This has been seen before. I think it must be an issue in the CircuitPython loader. Follow the instructions in [RESET.md](https://github.com/dbisu/pico-ducky/blob/main/RESET.md)

THANKS SM BROOOOO, IT WORKEDDDDD, HAPPY ASS FCKK

dbisu, 2023-04-01:
Closing, issue resolved.

closed 2023-04-01 by dbisu