dbisu/pico-ducky #161: pico-ducky gets no IP in AP mode

danielw1990, 2023-04-02:
**Describe the issue**
Pico Pi W does not get an IP when set up in AP mode.   
Prints out: "🐍Wi-Fi: No IP"

I can connect to the AP as expected, but trying to access 192.168.4.1 times out.

**To Reproduce**
1) Set up pico-ducky.   
2) Plug in and wait for the aforementioned message.
3) connect to AP
4) Try to access 192.168.4.1 and wait for timeout

**Expected behavior**
According to the guide it should have the IP 192.168.4.1 and should give a response if accessed via port 80 by a we bbrowser.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Payload gets executed too, mode change via pins works. Everything else seems to work.

danielw1990, 2023-04-02:
I tried to connect to the AP with my desktop PC. And somehow there I get a response.

So the problem changes to:
No response if accessed via chrome on Android and strange message printed out that claims to have no IP.

responsivereality, 2023-04-06:
If you are still having trouble with this, I created an updated repository for the Raspberry Pi Pico W. It does not contain any changed code, I have merely reorganised the amazing code that Dbisu developed. https://github.com/responsivereality/Dbisu-Pico-W-Ducky

dbisu, 2023-04-07:
It is possible that your phone won't stay connected to the Pico W because of the internet checks built into Android.

closed 2023-05-07 by dbisu