dbisu/pico-ducky #162: payload

AkumaBiNoTori, 2023-04-06:
i did everything exactly but the payload wont work the pico ducky respond when i ground the pins mentioned and everything its even shown as a keyboard when i plug it but no matter what the payload wont work. i tried to download circuit python 8.0.0 as i was on 8.0.5 but same result

responsivereality, 2023-04-06:
Hi there. Are you using the Pico W or the regular Pico? If you are using the W, the default state of the Mass Storage Device is switched.

responsivereality, 2023-04-06:
> i did everything exactly but the payload wont work the pico ducky respond when i ground the pins mentioned and everything its even shown as a keyboard when i plug it but no matter what the payload wont work. i tried to download circuit python 8.0.0 as i was on 8.0.5 but same result

Hi there. Are you using the Pico W or the regular Pico? If you are using the W, the default state of the Mass Storage Device is switched.

AkumaBiNoTori, 2023-04-07:
Its a regular pico

MikeCVermeer, 2023-04-07:
Same issue here. Installed everything but the payload does not get executed. My pc sees the pico as a HID

dbisu, 2023-04-07:
Which pins do you have tied together? If you have Pin 0 (GPIO 0) tied to Pin 3 (GND), the payload will not deploy automatically on boot, per design.

AkumaBiNoTori, 2023-04-07:
Well i didnt tie any pins at first but then i tied gp4 and ground and the pin that hides the pico still no payload but the pico is hidden

dbisu, 2023-04-08:
Are you on the latest code or one of the releases?

AkumaBiNoTori, 2023-04-08:
Right now i am on 8.0.0 last time i was on 8.0.5 i get the same result in both

dbisu, 2023-04-08:
With the current code in the repo, this is how the GPIO works for showing or hiding the Pico as a USB device
Pico:
*   GP15 not connected == USB visible
*   GP15 connected to GND == USB not visible

Pico W:
*   GP15 not connected == USB NOT visible
*   GP15 connected to GND == USB visible

AkumaBiNoTori, 2023-04-08:
Yes when i connect it the usb becomes invisible but still the payload wont work

dbisu, 2023-04-08:
Please get some debug logs
On Windows, use PuTTY, connect to the debug serial port (commonly COM3, but could vary)

On Linux, use minicom (minicom -b 115200 -o -D /dev/ttyACM0, where ttyACM0 corresponds to your device)

AkumaBiNoTori, 2023-04-09:
i am having trouble using the PuTTY  software i connected it and i am running it from cmd idk which command to choose

AkumaBiNoTori, 2023-04-09:
nvm i figured it out

AkumaBiNoTori, 2023-04-09:
[image]
this is hwat i get when i use it it just stays like this

dbisu, 2023-04-10:
In your PuTTY terminal, hit Ctl+c then Ctl+d to restart the program.

AkumaBiNoTori, 2023-04-12:
okay here is my update 
[image]

dbisu, 2023-04-12:
The code cannot find the file payload.dd  That file needs to be in the root folder of your Pico.

AkumaBiNoTori, 2023-04-13:
i do have it in the root 
[image]

dbisu, 2023-04-13:
It appears you have file extensions hidden. Make sure the file is named payload.dd and not payload.dd.txt

AkumaBiNoTori, 2023-04-13:
let me try

AkumaBiNoTori, 2023-04-13:
[image]
it is .txt here but i cant change it

dbisu, 2023-04-13:
Open File Explorer (This PC) → Click on View option in upper menu to view Ribbon menu → Check the box next the File name extensions option.
You should then get the full file name if you try to rename it.

AkumaBiNoTori, 2023-04-13:
finally it worked
thanks so much for your help

AkumaBiNoTori, 2023-04-13:
just a small question if i want it to inject certain packets i should place those packets normally in the root of the pico or will it cause any problem?

dbisu, 2023-04-14:
Not sure what you mean by packets. This tool emulates a keyboard, not a network device.

AkumaBiNoTori, 2023-04-14:
didnt mean a packet i meant a file malware like can i use it as a small usb that has data and i can transfer this data and make my code run it

Plato13, 2023-04-14:
My payload is not working to. Anyone else can help me ?

Plato13, 2023-04-14:
[image]

[image]

dbisu, 2023-04-15:
> didnt mean a packet i meant a file malware like can i use it as a small usb that has data and i can transfer this data and make my code run it

Yes, there is around 1MB of space on the Pico.

AkumaBiNoTori, 2023-04-15:
can i add an sd card module to it ? or will it cause problems?

dbisu, 2023-04-15:
Pi Pico boards don't support sd cards directly. Adding sd card support to the pico is outside the scope of this project.

AkumaBiNoTori, 2023-04-15:
okay thank you so much for your help

closed 2023-04-15 by AkumaBiNoTori