dbisu/pico-ducky #164: Wrong Case of letter or missing letters

PentestGuru1956, 2023-04-12:
when injecting text the letter case is reversed or the upper case letters are missing. happens about six of 10 times. making it hard  to troubleshoot scripts

pick a script and run it over and over to make sure it functions properly
ex:
cd %USERPROFILE% & netsh wlan show profiles | findstr "All" > a.txt
SMTPServer = 'smtp.gmail.com'
result:
cd %userprofiles% & NETSH WLAN SHOW PROFILES | FINDSTR "aLL" > A.TXT
--- so findstr doesn't work
-or-
cd %% & netsh wlan show profiles | findstr "ll" > a.txt
erver = SMTP.GMAIL.COM''
--- all capital letters are not injected

tried v7.3.3, v8.0.0, v8.0.3, v8.0.4 and v8.0.5

expected to output line as typed. Any help appreciated.

dbisu, 2023-04-15:
Please check your keyboard layout. The code defaults to a US keyboard.

PentestGuru1956, 2023-04-17:
Found the Problem , when it happens the caps lock is on.

closed 2023-04-17 by PentestGuru1956