dbisu/pico-ducky #165: Payload not running

Dezmur77, 2023-04-12:
I have tried to install pico-ducky on three different pico boards. All of them result in the LED no longer flashing and not giving any output. I've followed the guide exactly and used the flash_nuke firmware to reset and try again on each board. I've tried different self-written payloads and preset payload templates. When booting into putty and reloading, this is the output I'm given:

[image]

dbisu, 2023-04-13:
Your Pico is running CircuitPython 6.3.0. Please update to 8.0.x to run the latest pico-ducky code.

ji9-alt, 2023-04-13:
same here i was following a guide but the pico is not running any payload, I'm using ubuntu I'm not sure if that's affecting it

dbisu, 2023-05-20:
Closing, no activity.

closed 2023-05-20 by dbisu