dbisu/pico-ducky #168: Payload not running

Plato13, 2023-04-14:


Plato13, 2023-04-14:
> 

[image]

[image]

Can anyone help me ? I think finish everything but I couldnt have done.

clackes, 2023-04-15:
check in code.py

Plato13, 2023-04-15:
> check in code.py
> 

I cannot open code.py with pyhton only 
I open with thonny.

Plato13, 2023-04-15:
[image]

Plato13, 2023-04-15:
[image]

dbisu, 2023-04-15:
The code will not resolve you try to run it on your computer. It is made for CircuitPython running on the Pico.
Collect the output from the debug serial.
On Windows, use PuTTY, connect to the debug serial port (commonly COM3, but could vary)

On Linux, use minicom (minicom -b 115200 -o -D /dev/ttyACM0, where ttyACM0 corresponds to your device)

dbisu, 2023-05-20:
Closing, no activity.

closed 2023-05-20 by dbisu