dbisu/pico-ducky #176: Payload Not Working (COM3 Debug Output empty)

Whiteflame362, 2023-05-05:
I followed the steps and the payload isn't running. I plugged into the COM3 serial debug port to see the output and nothing prints or appears when I plug the Pico in and out. The LED is pulsating and I don't know what that means. Do I have to ground the payload I want before I attempt to use my Pico, or does it automatically run the payload.dd file?
[image]
[image]
[image]

dbisu, 2023-05-07:
It will automatically select payload.dd if no pins are selected.
You may need to check with Device Manager which as to which COM port it enumerated. COM3 was just an example.

dbisu, 2023-05-29:
Try using one of the releases if you are still having problems.
Closing, no activity.

closed 2023-05-29 by dbisu