dbisu/pico-ducky #179: Rasberry pi pico w issue with the payload runing

Crazzzzyyyy223, 2023-05-17:
Im having some issue getting the payloads to run on my raspberry Pico w i copy all the stuff in the places and put secrets.py in the root and put a payload.dd with a payload and nothing happens at all and then when i unplug it it doesn't show back up and then i have to reset my Pico.

dbisu, 2023-05-17:
Please post data from the debug serial port.
As for the device not showing up, by default with this code, the Pico W will not enumerate as a USB drive without being in USB mode.
[USB Mode](https://github.com/dbisu/pico-ducky#usb-enabledisable-mode)

Crazzzzyyyy223, 2023-05-17:
how do i check the debug cereal port

dbisu, 2023-05-17:
On Windows, use PuTTY, connect to the debug serial port (commonly COM3, but could vary)

On Linux, use minicom (minicom -b 115200 -o -D /dev/ttyACM0, where ttyACM0 corresponds to your device)

Crazzzzyyyy223, 2023-05-17:
[image]
my port is empty and the payload is still not working

Crazzzzyyyy223, 2023-05-17:
NVM I GOT I TO WORK JUST HAD TO AUTO  RELOAD

Crazzzzyyyy223, 2023-05-17:
THANK YOU SO MUCH

closed 2023-05-17 by Crazzzzyyyy223