dbisu/pico-ducky #180: Broken pipe error 32

akshayb72, 2023-05-20:
**Describe the bug**
I modified the code to only work with Pico W

**Modified Code**
code.py

import supervisor

import time
import digitalio
from board import *
import board
from duckyinpython import *
if(board.board_id == 'raspberry_pi_pico_w'):
    import wifi
    from webapp import *

# sleep at the start to allow the device to be recognized by the host computer
time.sleep(.5)

def startWiFi():
    import ipaddress
    # Get wifi details and more from a secrets.py file
    try:
        from secrets import secrets
    except ImportError:
        print("WiFi secrets are kept in secrets.py, please add them there!")
        raise

    print("Connect wifi")
    #wifi.radio.connect(secrets['ssid'],secrets['password'])
    wifi.radio.start_ap(secrets['ssid'],secrets['password'])

    HOST = repr(wifi.radio.ipv4_address_ap)
    PORT = 80        # Port to listen on
    print(HOST,PORT)

# turn off automatically reloading when files are written to the pico
#supervisor.disable_autoreload()
supervisor.runtime.autoreload = False

if(board.board_id == 'raspberry_pi_pico'):
    led = pwmio.PWMOut(board.LED, frequency=5000, duty_cycle=0)
elif(board.board_id == 'raspberry_pi_pico_w'):
    led = digitalio.DigitalInOut(board.LED)
    led.switch_to_output()

progStatus = False
progStatus = getProgrammingStatus()
print("progStatus", progStatus)
if(progStatus == False):
    '''
    print("Finding payload")
    # not in setup mode, inject the payload
    #payload = selectPayload()
    #print("Running ", payload)
    #runScript(payload)

    print("Done")
    '''
else:
    print("Update your payload")

led_state = False

async def main_loop():
    global led, button1
    wifi_initialized = False

    button_task = asyncio.create_task(monitor_buttons(button1))
[... 536 lines cut ...]

        if(button1Released):
            if(button1Down):
                # Run selected payload
                payload = selectPayload()
                print("Running ", payload)
                runScript(payload)
                print("Done")
            button1Down = False

        await asyncio.sleep(0)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
The code should run normally but shows broken pipe error 32

**Screenshots**

[image]

dbisu, 2023-05-20:
Have you compared your code with the working code from the repo?

akshayb72, 2023-05-20:
Yes for the most part it worked

akshayb72, 2023-05-20:
the main repo code works for both but i modified it so it would only work on pico wifi boards

akshayb72, 2023-05-21:
> Have you compared your code with the working code from the repo?

the main repo code works for both but i modified it so it would only work on pico wifi boards

Yes for the most part it worked but when code is run via webapp.py interface ,it works for sometime and shows broken pipe error 32 and i commented runScript in code.py.filenumber runscricpt function is commented out to avoid payload.dd causing broken pipe error

dbisu, 2023-05-29:
Debugging modified code is out of scope.
Is there a reason to not use the base code that supports both types of boards?

closed 2023-06-24 by dbisu

akshayb72, 2023-07-22:
Thanks finally got it working