dbisu/pico-ducky #181: my pico dcì

Ro7yt, 2023-05-21:
**Describe the issue**
A clear and concise description of the symptoms

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

Ro7yt, 2023-05-21:
when i order the pico ducky to write a link or any other special caracter it cange them. How do i fix this?

dbisu, 2023-05-21:
Have you ensured that you are using the correct keyboard layout? The default setting is for a US keyboard. If you have non-US keyboard settings, you will need to modify the changing keyboard layout instructions [here](https://github.com/dbisu/pico-ducky#changing-keyboard-layouts)

dbisu, 2023-05-29:
There is now a pre-packaged release that supports Pico W. Try one of those with the appropriate keyboard layout.

dbisu, 2023-06-24:
Closing, no activity.

closed 2023-06-24 by dbisu