dbisu/pico-ducky #185: plz help i have a raspberry pi pico but its not running payload

h54t45tv45, 2023-06-04:
i am on mac i have a windows school computer but i cant install python on it, i have done many tuts but non of them work

Thanoulimefonazoune, 2023-06-21:
Hi,
unfortunately I have Mac and I couldn't do it either but I found a windows laptop and did it it doesn't need to have python just follow the instructions meaning download the necessary files and dragging and dropping in rpi pico

h54t45tv45, 2023-06-21:
It does not work

ons. 21. jun. 2023 22.21 skrev Thanoulimefonazoune ***@***.***

Thanoulimefonazoune, 2023-06-21:
Try flashing the pico with the nuke flash and do the steps again sometimes it doesn't work also there is a video by network chuck do not follow it is for an older version follow all the steps in the READ Me.md file if there is a problème again try a different windows computer or a vm

h54t45tv45, 2023-06-21:
Ok, I'll check it out.

ons. 21. jun. 2023 22.35 skrev Thanos.

dbisu, 2023-06-24:
Your host or target computer do not need python. The python code runs directly on the Pico board.
the easiest way to get started is to follow the install instructions on the [Releases page](https://github.com/dbisu/pico-ducky/releases)

dbisu, 2023-09-17:
Closing, no activity.

closed 2023-09-17 by dbisu