dbisu/pico-ducky #186: Problem with de AP function

barsoom3q, 2023-06-09:
Hello.

I have a Raspberry Pi Pico W; when I plug the raspberry, it only runs the payload.dd, but does not create the AP, I don't know why.
The secrets.py it's OK.

I tried various solutions, but nothing happened.

I little help will be very helpful.

dbisu, 2023-06-24:
If possible, include debug serial data. On Windows, use PuTTY, connect to the debug serial port (commonly COM3, but could vary)

On Linux, use minicom (minicom -b 115200 -o -D /dev/ttyACM0, where ttyACM0 corresponds to your device)

shadowelite-sec, 2023-08-22:
hi i am also facing the same 

here the serial output 

```
Press any key to enter the REPL. Use CTRL-D to reload.        
r🐍Wi-Fi: No IP | code.py | 8.2.3Done
Starting Wifi
Traceback (most recent call last):
  File "code.py", line 82, in <module>
  File "asyncio/core.py", line 297, in run
  File "asyncio/core.py", line 261, in run_until_complete
  File "asyncio/core.py", line 246, in run_until_complete
  File "code.py", line 74, in main_loop
  File "code.py", line 27, in startWiFi
  File "secrets.py", line 1
IndentationError: unexpected indent
0;🐍Wi-Fi: No IP | 1@secrets.py IndentationError | 8.2.3
Code done running.
0;🐍Wi-Fi: No IP | REPL | 8.2.3
Adafruit CircuitPython 8.2.3 on 2023-08-11; Raspberry Pi Pico W with rp2040
>>> tepad
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'tepad' is not defined
```

shadowelite-sec, 2023-08-22:
It worked after changing the format to this

```
secrets = {
    'ssid': "SE",    
    'password': "elitesec" 
}
```

dbisu, 2023-08-23:
Is it possible on the first version there was a space before `secrets =`?
that would probably explain the error.

dbisu, 2023-09-17:
Closing, resolved.

closed 2023-09-17 by dbisu