dbisu/pico-ducky #187: Not working

Kaxpe4, 2023-06-09:
I did everything step by step, but my pico-ducky doesnt work.
The only way my device worked, is time when i used "Pico to bad usb converter", and it only worked once.

Payload should load, open notebook and type "Hello word", just like orginal payload

sorry for bad english

Sebastean2006, 2023-06-24:
Did you find out what was wrong? For me it didn't even work once.

Kaxpe4, 2023-06-24:
> Did you find out what was wrong? For me it didn't even work once.

No, I abandoned this project on my pico

Sebastean2006, 2023-06-24:
i found another git that worked for me:
https://github.com/OCEANOFANYTHINGOFFICIAL/RPI-PICO-TO-BAD-USB-CONVERTER

Sebastean2006, 2023-06-24:
If you still want to try

Kaxpe4, 2023-06-24:
> i found another git that worked for me:
> 
> https://github.com/OCEANOFANYTHINGOFFICIAL/RPI-PICO-TO-BAD-USB-CONVERTER

As I said in issue description I already used it. It worked a few times and it's dead.

dbisu, 2023-06-24:
If possible, include debug serial data.
On Windows, use PuTTY, connect to the debug serial port (commonly COM3, but could vary)

On Linux, use minicom (minicom -b 115200 -o -D /dev/ttyACM0, where ttyACM0 corresponds to your device)

dbisu, 2023-09-17:
closing, no activity.

closed 2023-09-17 by dbisu