dbisu/pico-ducky #188: String in start menu too long

TheJuicePapi, 2023-06-10:
It seems that pretty much any of the remote access tools when being used write the string code into the start menu on my windows 10 and most of the codes are so long they do not finish writing and the script stops there. I have yet to get a single working reverse shell working from these on my windows 10 using my kali linux machine as a listener. 

 I will follow the exact steps that are listed but the results always end with not enough room for the longest string to be typed in the bottom left search bar.

SENLINE232, 2023-06-15:
Use cloning a raw repository from github. Which will be the code you need

dbisu, 2023-06-24:
How long of a string are you using? It is possible you are hitting a command shell limit. There isn't a limit in the pico-ducky code, as long as it is on one line of the payload file.

dbisu, 2023-09-17:
Closing, not activity.

closed 2023-09-17 by dbisu