dbisu/pico-ducky #192: USB Disable not working on Windows

Thanoulimefonazoune, 2023-06-23:
I have read the instructions I have sorted pin 20 with 18 and on my Mac it shows as a usb device after all the code has run but in windows it shows in the start and the pop up explorer window interrupts the commands making the only way around it to delay all the commands in the beginning but then it is very clear that a new device has been connected to the PC

So I don't know if I am doing something wrong or is a bug.

dbisu, 2023-06-24:
Are you using a Pico or Pico W? On a Pico, shorting those pins with hide the USB storage device. On a Pico W, they enable the USB storage device.
I'll work to clean up the README to make that more clear.

Thanoulimefonazoune, 2023-07-08:
I have the normal not the W and I short these pins but it doesn't work it works on Mac but in windows it shows right away

dbisu, 2023-09-17:
If you are still seeing this problem, please include the contents of the boot logs.

dbisu, 2024-02-24:
Closing, no activity.

closed 2024-02-24 by dbisu