dbisu/pico-ducky #196: Error creating new scripts on W

XtramCZ, 2023-07-02:
**Describe the bug**
When i want to save a script on the Web UI, I get a crash

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Web UI
2. Click on New Script
3. Name the script and put some code in there
4. See error

**Expected behavior**
I expected it to save my script

**Screenshots**
-//-

**Debug info**
Traceback (most recent call last):
  File "asyncio/core.py", line 246, in run_until_complete
  File "webapp.py", line 265, in startWebService
  File "wsgiserver.py", line 165, in update_poll
  File "adafruit_wsgi/wsgi_app.py", line 70, in __call__
  File "webapp.py", line 211, in write_new_script
  File "webapp.py", line 138, in cleanup_text
  File "webapp.py", line 138, in <dictcomp>
MemoryError: memory allocation failed, allocating 4168 bytes

**Additional context**
-//-

Progreymer, 2023-08-01:
I am currently working on debugging this with the Thonny IDE and the current issue is that it cannot edit the drive.

**Debug Info**
{'scriptData': 'SCRIPT+HELLO+THERE', 'scriptName': 'Test.dd'}
```
Traceback (most recent call last):
  File "asyncio/core.py", line 246, in run_until_complete
  File "webapp.py", line 242, in startWebService
  File "wsgiserver.py", line 165, in update_poll
  File "adafruit_wsgi/wsgi_app.py", line 70, in __call__
  File "webapp.py", line 198, in write_new_script
RuntimeError: Cannot remount '/' when visible via USB.
```

Progreymer, 2023-08-02:
You're right. For my purposes I need USB mode enabled so I just deleted the drive mount lines in ```create_new_script()``` and added a try-catch. It's not 100% safe but it works for me as I have backups.

dbisu, 2023-08-09:
If USB Mass Storage mode is enabled, then the webapp cannot write to the internal flash. this is a limitation of circuitpython.

As for the original issue, do you have an example script that causes the problem? I'll try to take a look at this issue in the next couple of weeks.

XtramCZ, 2023-08-09:
Nah, I haven't noticed this happening with a particular script, but it happens more with longer ones

dbisu, 2024-12-25:
I've been able to reproduce this error. Trying to find a working solution.