dbisu/pico-ducky #199: Pico ducky payload is not executed after rebooting the connected android device

Karthiksgowda, 2023-07-21:
We are trying to automate some actions on an android device after factory resetting the android device. But after the android device is rebooted the pico SSID is not listed in the wifi network tab and the payload.dd is not being executed. tried with increasing the sleep in code.py. 

# sleep at the start to allow the device to be recognized by the host computer
time.sleep(120)

Expectations: After factory resetting the android device the wifi ssid should be listed and payload.dd should be executed

**Note : The pico flash light starts flashing 2 times after android device reset but normally it flashes once per second**

dbisu, 2023-08-23:
If possible, include debug serial data. On Windows, use PuTTY, connect to the debug serial port (commonly COM3, but could vary)

On Linux, use minicom (minicom -b 115200 -o -D /dev/ttyACM0, where ttyACM0 corresponds to your device)

dbisu, 2024-02-24:
Closing, no activity.