dbisu/pico-ducky #200: Feature request not BuG

Kronos2308, 2023-08-01:
## This week I was bored and among all your project it caught my attention how a small board can do all that.
## I'm using an RP2040 Zero because it's small.
## I made some changes to the code to better adapt to the device itself, and change the payload to software, we have a flash, it's better to use it XD
## In any case, I wanted to leave the public code in case someone is interested or wants to do something with it.

Saludos

### [https://github.com/Kronos2308/pico-ducky](https://github.com/Kronos2308/pico-ducky)

Kronos2308, 2023-08-18:
[image]

dbisu, 2023-08-23:
That is a neat little board.

Kronos2308, 2023-08-24:
Works like this
[image]
also i fix the code to be hable to chsnge paylodas usig a power bank

dbisu, 2023-10-17:
Closing as completed.
Thanks for sharing this project.

closed 2023-10-17 by dbisu