dbisu/pico-ducky #203: cant copy the .UF2 file

redshift-cyber, 2023-08-14:
**Describe the issue**
So i did all the steps before and it worked. Then the next day i deleted all the files off the pi and wanted to do the same thing as before. This time however the pico is alr named as "CIRCUITPY" and whenever i copy the .UF2 file i get an error saying there isnt enough storage space.

**To Reproduce**
Copy the .UF2 file to the alr named circuitpy it gives an error saying "Not enough storage space"

**Expected behavior**
I expected the .uf2 file to get copied there where the pi would reboot.

**Screenshots**
[image]

dbisu, 2023-08-15:
If you want to update the uf2 version, you need to hold the reset button on boot of the Pico. You should then see the Pico show up as RPI-RP2.
You can then copy the new uf2 file.

dbisu, 2023-09-17:
Closing, resolved

closed 2023-09-17 by dbisu