dbisu/pico-ducky #204: Problem with brackets "{"

ivoiot, 2023-08-25:
i faced a problem with italian layout setting on my raspberry pi pico W. when i execute some scrit with the simbol '{' or '}' it get stucked. any suggestion to fix it?

dbisu, 2023-09-17:
I don't have this layout, so hard to debug. My suggestion would be use something like showkey to determine if what the script is sending matches what you see when you press the same key on the keyboard.

dbisu, 2023-09-17:
It looks like this could be due to older libraries in the release bundles. there is an update from Neradoc with curly bracket support on Italian keyboards.
https://github.com/Neradoc/Circuitpython_Keyboard_Layouts/releases/tag/20221209
You can try updating the libraries on your ducky.

dbisu, 2023-10-18:
Have you tried updating to new libraries from Neradoc?
If it works, I'll make sure to update these libraries when I make the next release.

dbisu, 2024-12-21:
Try updating to the v3.0 release. The WIN_IT libraries have been updated to the latest versions.