dbisu/pico-ducky #205: Increase DELAY more 50000ms, stops the WSGI server

Milindsoni08, 2023-08-28:
Increase DELAY more 50000ms, stops the WSGI server and i am unable to access 192.168.4.1
payload.dd
DELAY 50000

if i keep DELAY < 50000ms it works like a charm

Karthiksgowda, 2023-09-05:
The above mentioned solution is not working @dbisu

dbisu, 2023-09-06:
Yeah, I'm still working on a solution. Like I said, might take a bit to figure out the right way to refactor the code.

ihsaank, 2023-09-12:
Hi, I'm only able to connect to the pico w for literally 30 seconds then the web server stops loading. Unless I forget the SSID and reconnect then it only works for a few seconds again. Do you know what the issue is?

dbisu, 2023-09-17:
> Hi, I'm only able to connect to the pico w for literally 30 seconds then the web server stops loading. Unless I forget the SSID and reconnect then it only works for a few seconds again. Do you know what the issue is?

@ihsaank please open a separate issue for this problem.

dbisu, 2023-09-17:
For the original problem, I've created a test branch, delay-refactor, with proposed changes to fix this issue.

ihsaank, 2023-09-17:
Is the original issue the same as the problem I’ve stated here?

dbisu, 2023-09-17:
Are you trying to use long delays and seeing the webserver stop?
A log dump from the debug serial would be helpful.

ihsaank, 2023-09-17:
Oh okay sorry no it’s a different issue I’m facing. I’ve followed all the installation steps correctly so not sure if it’s a firmware issue. Once I’m connected to the SSID I’m only able to execute payloads only for a few seconds then the web server freezes up and stops loading and seems to show I can no longer connect and when I go to the wifi option it shows a cross across the wifi and ‘no internet connection’ underneath it

dbisu, 2023-09-17:
This is a different problem, please open a separate issue to track it.