dbisu/pico-ducky #206: Increasing Storage

Writepro, 2023-09-20:
am interested in how you can add storage if its possible to the pico so u can copy files into it maybe an sd or micro sd card

dbisu, 2023-09-20:
Take a look at this project.
https://www.instructables.com/Raspberry-Pi-Pico-Micro-SD-Card-Interface/

Ethank1354, 2023-10-02:
I gave this a try. I found that because the project mentioned above is micropython not circuitpython, it doesn't work one for one as circuitpython doesn't have the machine library. I have not looked into setting it up using the circuitpython equivalents yet. I ended up using sdcardio for this. The largest problem I found is that all fo the documentation mentions using board.SPI, board.SCK and the same for mosi and miso, but those don't exist on circuitpython for the pi pico, so you need to specify the pins by board.GP13 for example. GP10 and 11 are also in use for specifying other payloads, so if you wire it exactly the same as the project above, you will need to change the pins in duckyinpython.py

dbisu, 2023-10-17:
Closing as completed.

closed 2023-10-17 by dbisu