dbisu/pico-ducky #210: The Website Is Not Working

notbergen, 2023-10-03:
the website that you control the pico duckey is not letting me edit or create new scripts

ive tried nuking the pico and change the wifi name and password

im wanting to edit and make new scripts

every time i try to edit or make new scripts/payloads i name then in order from payload.dd payload2.dd and so on the website crashes and says it cant be reached even when im on the pico w's network please help thank you

Ethank1354, 2023-10-03:
That happens when the pico w is also mounted as a mass storage device, I saw in another issue that the web server can't write to the internal storage when mass storage mode is activated

notbergen, 2023-10-10:
how do i unmount it ?

Ethank1354, 2023-10-11:
Disconnect pin 18 from pin 20, in the main page, there is a section just under setup mode about it

dbisu, 2024-12-25:
While debugging another issue, I've run across a similar issue. I'm researching a solution. I think it might have to do with some I/O calls being blocking calls, causing the server thread to time out.