dbisu/pico-ducky #211: After doing all that setup raspberry pi pico doesn't recognize as hid device and not run commands simply open as storage device.

Vivu5772, 2023-10-09:
**Describe the issue**
A clear and concise description of the symptoms

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Debug info**
If possible, include debug serial data.
On Windows, use PuTTY, connect to the debug serial port (commonly COM3, but could vary)

On Linux, use minicom (minicom -b 115200 -o -D /dev/ttyACM0, where ttyACM0 corresponds to your device)

**Additional context**
Add any other context about the problem here.

Vaneer420, 2023-10-17:
also experiencing this issue

dbisu, 2023-10-17:
Please see [USB Enable/Disable](https://github.com/dbisu/pico-ducky#usb-enabledisable-mode) section.
By default, a Pico W will not show up as a USB mass storage device to enable the web service to use the storage.

tranquillechat, 2023-12-24:
Hello,
Same wird issue, follow the quickstart version 2.0 and 1.4, or the complete guide, everything copy fine, but i have no "hello world" payload working, deconnect/reconnect, nothing.

Any ideas of any mistapes ?!

closed 2024-02-15 by dbisu