dbisu/pico-ducky #216: Stopping AP is not supported in asyncio

mattiavivi, 2023-11-05:
good morning, after having warned my raspberry pico w it gives me an error that asyncio.run(main_loop()) does not support access point mode of the wifi connection with wifi.radio.start_ap(secrets['ssid'],secrets['password' ])

`async def main_loop():
    global led,button1

    button_task = asyncio.create_task(monitor_buttons(button1))
    
    pico_led_task = asyncio.create_task(blink_pico_w_led(led))
    print("Starting Wifi")
    #startWiFi()

    import ipaddress
    # Get wifi details and more from a secrets.py file
    secrets = {
    'ssid' : "-----",
    'password' : "----"
    }

    print("Connect wifi")
    wifi.radio.start_ap(secrets['ssid'],secrets['password'])
    
    # Port to listen on
    print("conesso", wifi.radio.ipv4_address_ap)

    print("Starting Web Service")
    webservice_task = asyncio.create_task(startWebService())
    await asyncio.gather(pico_led_task, button_task, webservice_task)
    
asyncio.run(main_loop())`

dbisu, 2023-12-26:
due to some limitations in circuit python, you cannot just Ctl-D to restart the pico-w. You will need to unplug it from your computer and restart it.

dbisu, 2024-01-28:
Closing, this is an issue in CircuitPython, not specific to this project.

closed 2024-01-28 by dbisu