dbisu/pico-ducky #218: I can't change my keyboard layout, what am I doing wrong

wsrb77, 2023-12-07:
I went by instructions but he still uses the us layout.
Pictures are attached

ADD the de-Layout
[image]

Change Layout 
[image]

dbisu, 2023-12-09:
Looks correct. what errors are you getting?

wsrb77, 2023-12-11:
I didn't have any errors, everything worked fine, except that it used the US keyboard layout again when executing the payload.
I found the preconfigured file in German, and it worked regardless. Thank you anyway.

dbisu, 2023-12-11:
Closing as resolved.

closed 2023-12-11 by dbisu

exequze, 2024-02-23:
i had the same issue, you have to add "adafruit_hid." to the keayboard layout for other languages for it to work. maybe add this to the code :)