dbisu/pico-ducky #219: Can used on victimPC even if it's off

maroc46, 2023-12-24:
Can I used the pico grabber on a shutted down pc to grab credentials? 

Or is there a another way, I think I can used a geparted, clonezilla to copy and See with linux the windows files 

Is there a way to grad files on shutted down pc or laptop ??

dbisu, 2023-12-26:
No, this project acts like a HID, specifically, a keyboard. It can only do things you can do with a keyboard.

maroc46, 2023-12-26:
Thank you for the fast awnser. 

I seen a doucumentary on youtube about a scammer.

https://youtu.be/4cj--G4Nc4E?si=hB5ElrWX2xgjA11b

In 10:38 he talk about the pi pico and that it's can be used as a (hid) to act like mouse and keyboard 
In 10:38 he awnser a question from the interviewer that asked him : so he can grab his files from his laptop with a pico

And he say yes ;even if it's off. 

*I don't know what a code he "use" for his pico 

For me the logical thing that be can Is only the pico-ducky 

But iam now think that he don't realy know what he says. 

Thank you

dbisu, 2023-12-26:
Yeah, that was a lot of bluster from him. Pico-ducky definitely does not work if the computer is off, as it just acts like a keyboard. So the computer needs to be running to take in keystrokes.

dbisu, 2024-01-28:
Closing, answered.

closed 2024-01-28 by dbisu