dbisu/pico-ducky #220: not working

smrf356, 2023-12-29:
when I plug the usb in to my chrome book and my pc it comes up as a usb device but it dose not play my payload. 
I redownloaded the Pico ducky software 3 or 4 times but it dose not work

dbisu, 2023-12-29:
Are you using a Pico or Pico-W?
Please include debug serial data.
On Windows, use PuTTY, connect to the debug serial port (commonly COM3, but could vary)
On Linux, use minicom (minicom -b 115200 -o -D /dev/ttyACM0, where ttyACM0 corresponds to your device)

smrf356, 2023-12-29:
I am sorry I don't want to waste your time but I have know idea how to use puddy I just got into it 5 days ago
(I am using just a Pico not a Pico-w)

dbisu, 2023-12-29:
Have you tried using one of the preconfigured release bundles?
You can find them [here](https://github.com/dbisu/pico-ducky/releases)
They tend to work best for new users.

smrf356, 2023-12-29:
[image]
[image]
I followed it and it still did not work is this what it is supposed to look like?

dbisu, 2023-12-29:
You are missing most of the recent files. What release version did you use? Version 2.0 is the latest and currently supported version.
https://github.com/dbisu/pico-ducky/releases/tag/v2.0

smrf356, 2023-12-29:
can you show me a pic of what it is meant to look like

dbisu, 2023-12-29:
You should have 6 .py files in the root of the Pico.
Make sure you are following the instructions in INSTALL.txt or in the INSTALL section at https://github.com/dbisu/pico-ducky/releases/tag/v2.0
This should walk you through how to get it working.

smrf356, 2023-12-29:
[image]
like this?

smrf356, 2023-12-29:
i got it to work
thank you so much for the help sorry I am kind of dumb

dbisu, 2023-12-29:
Good deal.
Hope you learned a bit about using the Pico.

Closing as resolved.

closed 2023-12-29 by dbisu