dbisu/pico-ducky #222: When i plug the pico in nothing happens i followed all the steps and its still not working.

Zachouile, 2024-01-16:
**Describe the issue**
A clear and concise description of the symptoms

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Debug info**
If possible, include debug serial data.
On Windows, use PuTTY, connect to the debug serial port (commonly COM3, but could vary)

On Linux, use minicom (minicom -b 115200 -o -D /dev/ttyACM0, where ttyACM0 corresponds to your device)

**Additional context**
Add any other context about the problem here.

Zachouile, 2024-01-16:
I followed all the steps and everything was good until when i plugged it in nothing happened i dont know what to do pls help me.

dbisu, 2024-01-16:
Are you using a Pico or Pico W?
Have you tried using one of the Release bundles for your keyboard language?

dbisu, 2024-01-18:
I would recommend trying to use one of the release bundles. You can find them [here](https://github.com/dbisu/pico-ducky/releases).

dbisu, 2024-02-15:
closing, no activity.

closed 2024-02-15 by dbisu