dbisu/pico-ducky #224: I followed all the steps and moved payload.dd to CIRCUITPY as the last step but nothing happened.

Blinn92, 2024-01-24:
I followed all the steps and moved payload.dd to CIRCUITPY as the last step but nothing happened. What I'm wondering is, what is the cause of the problem and how can I fix it so that it works normally?

dbisu, 2024-01-25:
If you are using a Pico, the payload runs only on boot. After copying your payload file (payload.dd), unplug the device and put into the target computer.

jepunband, 2024-02-23:
hello @dbisu , i completed all the steps. Now, when i plug the pico in all i got on windows like in the image below, i used the basic payload.dd to test it out.

[image]

dbisu, 2024-03-07:
From these pictures, seems like your delay isn't long enough. It looks like it is trying to type the Hello World before getting notepad open.

dbisu, 2024-09-21:
Closing, no activity.

closed 2024-09-21 by dbisu