dbisu/pico-ducky #230: Kind of a dumb question (setup mode)

devarxify, 2024-02-06:
Hello. Is there any possible way of entering setup mode without having to connect GP0 and GND? My jumper wires aren't think enough and therefore it does not work.

Thanks in advance.

dbisu, 2024-02-07:
Setup mode is so you can update the scripts without them running against your host computer. Running a jump or using a switch are the easiest ways to accomplish that.
You could modify the code to reverse the logic (always in setup mode), update your scripts, and then change it back. That is more work and if not careful could still cause the script to run against your computer.

devarxify, 2024-02-09:
@dbisu

dbisu, 2024-02-10:
change the code in [this](https://github.com/dbisu/pico-ducky/blob/main/duckyinpython.py#L125) function.

devarxify, 2024-02-10:
And how do you think I could make the Pico detect that I want to go on setup mode? I don't know if I explain myself correctly...

dbisu, 2024-02-10:
Have you actually looked at the code? it is fairly straight forward on how it figures out if the device is in setup mode or not.

devarxify, 2024-02-11:
Of course, it figures out if the device is in setup mode or not if the pins are connected, but I'm not really sure how else I could tell it that I want to go on setup mode

dbisu, 2024-02-12:
Not sure I'm following the problem.

devarxify, 2024-02-14:
Let me explain it better: I want to go onto setup mode without having to connect the two pins. I need a way to tell the Raspberry Pi Pico that I want to go on setup mode without having to connect the pins.

dbisu, 2024-02-14:
OK. I've pointed out the code the decides if it is in setup mode or not. You should be able to modify that to force it into setup mode and out again.

devarxify, 2024-02-15:
The issue is not with modifying the code, it's to give the code the signal that I want to go on setup mode

dbisu, 2024-02-16:
So you asked about my second suggestion, which was to modify the code to be in setup mode, add your payload, then modify it back to not be in setup mode.
Buttons or switches would be the other way, which are already supported by the code. You would need added hardware.

devarxify, 2024-02-16:
Alright, thank you very much.

closed 2024-02-16 by devarxify