dbisu/pico-ducky #231: Don't understand guide

HUnde125, 2024-02-07:
The guide says i should copy some files from "lib" but in  the newest version i downloaded one file isn't included and maybe you can make the guide more easy

dbisu, 2024-02-08:
Which file is missing?
Also, the quickest start is to use the release bundles [here](https://github.com/dbisu/pico-ducky/releases)

HUnde125, 2024-02-08:
Hey,
So I only need to download fe. the pico-ducky-v2.0-win-de file and paste the lib folder into my PICO and replace all files?

I tried the Full install and when I downloaded "adafruit-circuitpython-bundle-8.x-mpy-YYYYMMDD.zip", I couldn't find the "adafruit_debouncer.mpy " file

dbisu, 2024-02-09:
Yes, if you use the release bundle, all the needed files are included.

I check the latest adafruit bundle and I still see adafruit_debouncer.mpy in the lib folder.

dbisu, 2024-03-07:
Closing, no activity.

closed 2024-03-07 by dbisu