dbisu/pico-ducky #232: Pico W "New Script" Malfunction

FeliXthEca8, 2024-02-08:
Whn clicking the "Submit" button on the NewScript section, the page will crash and you cannot accsess this, or any other pages.

The server should be accepting the new script. This could be a hardware issue though it is not likely. 
Editing scripts still works.

The bug is major and stops all functionality to the PIco W unless it is unplugged and replugged. 

I did try to "Flash Nuke" the pico and reinstalled each script but the bug was still active. 

It might have to do with my CIRCUTPY version, ath this time i was using v8.2.9 (The latest LTS at this time)

This was made on 2/7/2024 at 9:15 PM.

FeliXthEca8, 2024-02-08:
*If you need serial logs please let me know

dbisu, 2024-02-09:
Yes, serial logs would be helpful. I'll try to look over the weekend.

BS-DevSec, 2024-02-20:
I had the same issue, but while trying to do that programmatically with a Python script.

```
requests.post('192.168.4.1', {'script name': 'test.dd', 'scriptData': 'TESTSTRING'})
```

psamx, 2024-04-11:
Have you enabled usb drive?