dbisu/pico-ducky #233: typo

Blinn92, 2024-02-09:
When I plug the USB into the computer, it works normally but it has one problem: it prints incorrectly. 
I don't know why it was typed incorrectly.

example
The command that I wrote
GUI r
DELAY 500
STRING notepad
ENTER
STRING a b c d e f g h i j k l m n o p q r s t u v w x y z

After run paylond is finished
 j k l ; n o p a r s t u v z x y wbcdefghijklnopqrstuvwxyz

sample picture
[image]
[image]

dbisu, 2024-02-10:
are you using US keyboard or a different keyboard layout?

dbisu, 2024-09-21:
Closing, no activity.

closed 2024-09-21 by dbisu