dbisu/pico-ducky #235: ducky script isn't running

devraaj11, 2024-02-19:
**Describe the issue**
the ducky script isn't running when connecting the pico
it just shows up as a usb device
all files were copied according to the install guide

**Expected behavior**
payload.dd is supposed to open notepad

**Screenshots**
Files in pico

[image]
files in lib

[image]

payload file

[image]

dbisu, 2024-02-20:
A couple of questions. Which keyboard layout are you using? Do you have setup mode still enabled? the scripts only run on boot, so make sure you update your script, then unplug and replug your pico.

devraaj11, 2024-02-20:
> A couple of questions. Which keyboard layout are you using? Do you have setup mode still enabled? the scripts only run on boot, so make sure you update your script, then unplug and replug your pico.

keyboard is eng-us
step mode is off
tried updating it but nothing happened

dbisu, 2024-02-24:
Please use putty to get a debug serial trace.

dbisu, 2024-09-21:
Ensure you are using CircuitPython 8.x and not 9.x.
Closing.

closed 2024-09-21 by dbisu