dbisu/pico-ducky #236: malfunction error

Y0SH1zzzz, 2024-02-22:
**Describe the issue**
after rebooting with adafruit-circuitpython-raspberry_pi_pico-en_US-8.0.0.uf2 i cant open any file or transfer any file without the explorer window freezing and then windows gives me an error about malfunctioning drive

**To Reproduce**
do first step

**Expected behavior**
not this

**Additional context**
Add any other context about the problem here.

coder0107git, 2024-02-22:
What OS are you using?

Y0SH1zzzz, 2024-02-22:
I am using windows Windows

dbisu, 2024-02-24:
Try following the reset instructions [here](https://github.com/dbisu/pico-ducky/blob/main/RESET.md).

Y0SH1zzzz, 2024-02-24:
no i still get the same error and nothing gets written to the pico, this only happens when placing the circuitpython file and it has rebooted

dbisu, 2024-02-25:
Have you tried a newer circuit python uf2?
You can find the latest here [https://downloads.circuitpython.org/bin/raspberry_pi_pico/en_US/adafruit-circuitpython-raspberry_pi_pico-en_US-8.2.10.uf2](https://downloads.circuitpython.org/bin/raspberry_pi_pico/en_US/adafruit-circuitpython-raspberry_pi_pico-en_US-8.2.10.uf2)

Y0SH1zzzz, 2024-02-25:
it only transfers around 60kb before disconnecting

dbisu, 2024-03-07:
Have you tried doing a reset and then trying to load the circuitpython uf2?
https://github.com/dbisu/pico-ducky/blob/main/RESET.md

dbisu, 2024-09-21:
Closing, no activity.

closed 2024-09-21 by dbisu