dbisu/pico-ducky #237: How to put Pico Pi back to storage mode?

jepunband, 2024-02-23:
Hello, i have setup the Pico Pi W to work as rubber ducky as described in the readme. Now i want to add other payload and also modify/edit files inside Pico Pi W. How do i do this since it only detects as HID. I want to be able to transfer files. 

Anyone can help me on this?

dbisu, 2024-02-24:
See the instructions[here](https://github.com/dbisu/pico-ducky?tab=readme-ov-file#usb-enabledisable-mode).

dbisu, 2024-02-25:
Storage mode doesn't require pushing the BOOTSEL button on the Pico. Holding that button will put it into bootloader mode. In that mode it behaves like you are seeing.
The linked instructions how which pins to connect to make storage mode work on both Pico and Pico W. They behave slightly differently so make sure you looking at the right directions.

jepunband, 2024-02-25:
Ok thanks i managed to get the jumper cable and use that to get into the storage mode to update the files.

dbisu, 2024-03-07:
Closing, resolved.

closed 2024-03-07 by dbisu