dbisu/pico-ducky #238: How to get Pico to only execute payload on "RUN"

jepunband, 2024-02-23:
Hello,

Have setup Pico Pi W and i wanted to make sure once connected to PC that the payload does not execute immediately. I only want it to execute once in connect to its Wi-Fi, on the webpage itself and execute/run payload. 

How do i do this ?

dbisu, 2024-02-24:
Put the device in setup mode. it will not auto run the payloads, but will allow them to be run from the webpage.

dbisu, 2024-02-25:
The best option would be to have a PCB created with switches. You can look [here](https://github.com/dbisu/pico-ducky/tree/board-design/board-design) for some CAD files for a board I created. You can then make an enclosure around that board.
You can also look at [this project](https://github.com/iot-pwn/defcon31-ducky) for another board I had created around the Pico W.

psamx, 2024-04-11:
You can archive this naming your payload files anything other than payload.dd

dbisu, 2024-09-21:
Closing, no activity.

closed 2024-09-21 by dbisu