dbisu/pico-ducky #242: Pico-Ducky not executing script

stefan2jz, 2024-02-27:
**Describe the issue**
Ive done this with the help of the tutorials and websites and your webite multiple times . When i get the normal payload it doesnt execute anything . When i plug it in , it opens the file explorer only and doesnt go as a HID. Doesnt execute the script . Idk how to fix.  Im owning a Raspberry Pi Pico

Idk what to change , did everything like in the tutorial . The payload is : 

REM The next four lines open Notepad in Windows and type "Hello World!"
GUI r
STRING notepad
ENTER
DELAY 250
STRING Hello World!

**Screenshots**
These are the screenshots from the Pico 
[image]
[image]
[image]

Please help me asap!!

dbisu, 2024-03-07:
Please submit a debug serial trace of the issue.

dbisu, 2024-09-21:
Ensure you are using CircuitPython 8.x and not 9.x
Closing.

closed 2024-09-21 by dbisu