dbisu/pico-ducky #243: pico w not executing payload.dd after all steps

oferrr, 2024-02-28:
**Describe the issue**
A clear and concise description of the symptoms

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Debug info**
If possible, include debug serial data.
On Windows, use PuTTY, connect to the debug serial port (commonly COM3, but could vary)

On Linux, use minicom (minicom -b 115200 -o -D /dev/ttyACM0, where ttyACM0 corresponds to your device)

**Additional context**
Add any other context about the problem here.

oferrr, 2024-02-28:
the device manager pick the pico w as hid device but nothing is happening, only the led starting to flesh.

oferrr, 2024-02-28:
[image]
got that error while executing putty.exe -serial com3

oferrr, 2024-02-28:
I unplugged the pico completely and i got...the same error? wth

oferrr, 2024-02-28:
so, i was dumb. i wanst on the correct COM. ops

sillymonkey87, 2024-03-06:
Im having an issue where i did everything the guide told me to, and my computer detects the pico w as a hid device. but no matter what. it wont run the payload

dbisu, 2024-03-07:
Are you able to get a debug serial trace of the issue?

dbisu, 2024-09-21:
Ensure you are using CircuitPython 8.x and not 9.x
Closing.

closed 2024-09-21 by dbisu