dbisu/pico-ducky #246: Pico ducky not working with all steps done

Miranshkashyap, 2024-03-10:
After putting payload to pico nothing happens
[image]
Files in pico

[image]
Files in library

Please help

Miranshkashyap, 2024-03-10:
Payload
[image]

dbisu, 2024-03-10:
I don't see payload.dd in the file list of the first picture.
Have you tried started with the release bundle? You can find the current release [here](https://github.com/dbisu/pico-ducky/releases/tag/v2.0) along with instructions on how to install it.

Miranshkashyap, 2024-03-10:
Even after adding it, it doesn't works

Miranshkashyap, 2024-03-10:
It works now thanks

closed 2024-03-10 by Miranshkashyap