dbisu/pico-ducky #247: Large payload file problem

denyzkoo, 2024-03-12:
Hi

 If i use 30 commands payload it works as charm. Im using raspberry_pi_pico. 

But if I use payload with 500 lines it execute only frist line and dont enter other 499 comands. 

Is there any limitations? 

Actualy i need to use about 50 000 lines payload maybe more. Size of payload i need to use is 845 kB. 

Thank you

dbisu, 2024-03-13:
There isn't a direct limit. Please post debug serial logs. Instructions [here](https://github.com/dbisu/pico-ducky/blob/main/DEBUG.md).

dbisu, 2024-12-25:
For performance reasons, the code loads the whole file into memory. It is possible this is causing the system to run out of memory on larger scripts. I'll look at the trade-off of having the whole script in RAM versus loading one line at a time.