dbisu/pico-ducky #250: Variables and Loops Not Working

acsmith1986, 2024-04-11:
I have a Raspberry Pi Pico running Circuit Python and Pico-Ducky, I can get normal Basic ducky scripts to run such as:

STRING This is a test 
DELAY 1000
STRING One second delay etc

however, I seem to be having issues with variable and loops etc from ducky scripts. for example

VAR $WAIT = 1000
STRING This is a test 
DELAY $WAIT
STRING One second delay etc

this should run exactly the same as the first scrip in my understanding, but it gets to the "This is a test" line and stops, it does not seem to operate with the delay variable.

I have the similar issue with Loops. 

VAR $COUNT = 5
WHILE ($COUNT > 0)
[TAB]     STRING This message will repeat 5 times
[TAB]     $COUNT = ( $COUNT -1 )
END_WHILE

Both of the above examples I have both typed and copied from the HAK5 reference documentation and the scripts will not function.

Is there a limitation to running these scripts on a RPI that means it cant run the full ducky scripts? or is there something I am missing or not doing correctly?

TIA

dbisu, 2024-04-11:
This project isn't compatible with Ducky Script 3.0. Ducky Script 1.0 didn't include these features.

acsmith1986, 2024-04-30:
Fair enough, thanks for the response.

closed 2024-04-30 by acsmith1986