dbisu/pico-ducky #253: how can i reduce the speed of how fast each character is typed?

salemnaasan123, 2024-04-30:
how can i reduce the speed of how fast each character is typed?

dbisu, 2024-05-01:
probably the easiest way would be to have each character as its own STRING line. If you need more time, you can add DELAY lines between the STRING lines.

salemnaasan123, 2024-05-01:
Thanks for the feedback, but I am asking because I have like a text of 500 words for exempel that I want to be written, and it will take so long if I just put a delay between each one. My plan is that to make an random delay (from 50 ms to 100 ms) between each charchter. I have tried to modify the code a little but didn't work, I appreciate so much if you give me a solution.

salemnaasan123, 2024-05-05:
Do you have any idea?

salemnaasan123, 2024-05-05:
Never mind, Ive edit a bit on the code and now I have a random delay between every character.

dbisu, 2024-05-07:
ok, sounds good.
Closing, resolved.

closed 2024-05-07 by dbisu