dbisu/pico-ducky #255: Pico payload is not running

ProtoTypd1, 2024-05-04:
The Pico is not opening notepad, as expected

Lib folder
[image]
root folder
[image]
Payload
[image]

Serial debug using putty
[image]

I am using the payload that comes with the install. pls help

ProtoTypd1, 2024-05-05:
i just used the regular python modules and it worked

dbisu, 2024-05-07:
This could be caused if you loaded circuitpython 9.x instead of 8.x. I have not yet updated everything to ensure it works with 9.x.

ProtoTypd1, 2024-05-08:
ok thanks

dbisu, 2024-05-09:
Closing, resolved.

closed 2024-05-09 by dbisu