dbisu/pico-ducky #256: Question: After Payload run a Batch file

Alznex, 2024-05-06:
Is there a way to start running a Batch datei, because i have many script in them and dont want to work myself in this payload.dd.
I dont wannt it to be run over the payload, but rather in the python code, i tried it with os or subprocess, but i dont know if i did somethink wrong or oversaw somethink. pls Help.

dbisu, 2024-05-07:
Not sure I'm following the question.
the python code runs on the pi pico, not on the host computer, so os or subprocess calls will not call to the host computer.
You can run batch files, similar to how you would run notepad.exe in the example payload.

Alznex, 2024-05-07:
Thanks for replaying so fast, sorry that i diddnt make clear exactly what i am trying. So my batch file is on the raspberry and is supposed to execute befor or after the payload. Sadly i dont kno the mistake i did but trying to run it trough os or subprocess throw out some kind of error i guess because than the payload also did not start, i know that is has not somethink todo with the lib it self because if i let the batch file be executet after the payload, the payload starts and it looks fine exept that the batch is not executet.

dbisu, 2024-09-21:
Closing, no activity.

closed 2024-09-21 by dbisu