dbisu/pico-ducky #257: Other keyboards dosen't work for me

tommarekCZE, 2024-05-08:
**Describe the issue**
Cannot use other keyboards, when I uncomment
```
#from keyboard_layout_win_cz import KeyboardLayout
#from keycode_win_cz import Keycode

```
It basically on connect wont load (I commented the US layout)
With us layout it works (instead of numbers its butting these types of characters: ěščřžčš)

**To Reproduce**
Uncomment:
```
#from keyboard_layout_win_cz import KeyboardLayout
#from keycode_win_cz import Keycode

```
Comment:
```
from adafruit_hid.keyboard_layout_us import KeyboardLayoutUS as KeyboardLayout
from adafruit_hid.keycode import Keycode
```

**Expected behavior**
It would run the code with the czech keyboard

**Debug info**
None

**Additional context**
I setup everything correctly.
I tried even putting the "keycode_win_cz.mpy" and "keyboard_layout_win_cz.mpy" into the lib forlder and into the root folder, didn't helped it.

dbisu, 2024-05-09:
Have you tried the prebuilt release bundle?
[CZ bundle](https://github.com/dbisu/pico-ducky/releases/download/v2.0/pico-ducky-v2.0-win_cz.zip)

tommarekCZE, 2024-05-09:
That worked thanks! It turned out I didnt copied all libs, I folowed tutorial at repo page. But it turned out I need copy full folder from the czech budle. Thanks

dbisu, 2024-05-09:
No problem. Thanks for reporting back.
Closing, resolved.

closed 2024-05-09 by dbisu