dbisu/pico-ducky #258: ducky script  Not Run

indiahackercyber, 2024-05-10:
**Describe the issue**
hi i am dev and you best write steps i am follow but my raspberry pi pico also make pico ducky but i am make file "payload.dd" but not run any ducky script  or ducky payload i am try over 1000 + ducky scripts include  android,windows,linux but not any run i am you resolve my error or issue i am hop   

of my scrips i am try 1000+ but i am show only 5 

1 for android 

REM Android Web Browser Automation
DELAY 1000
GUI r
DELAY 500
STRING chrome
DELAY 500
ENTER
DELAY 1500
STRING https://www.example.com
ENTER
 
(1) 2 
DELAY 1000
GUI r
DELAY 500
STRING cmd
DELAY 500
ENTER
DELAY 1000
STRING cd /sdcard
DELAY 500
ENTER
DELAY 500
STRING mkdir my_directory
DELAY 500
ENTER
DELAY 500
STRING cd my_directory
DELAY 500
ENTER
DELAY 500
STRING echo This is a test message > test.txt
DELAY 500
ENTER
DELAY 500
GUI q

for linux 

1(1) 

DELAY 1000
GUI
DELAY 500
STRING terminal
DELAY 500
ENTER
DELAY 1000
STRING sudo apt update && sudo apt upgrade -y
DELAY 500
ENTER

and more.............

dbisu, 2024-05-11:
Please include information from the debug serial port. I cannot help without more information on the issue.

dbisu, 2024-09-21:
Closing, no updates. Ensure you are not trying to use CircuitPython 9.x and only using 8.x for now.

closed 2024-09-21 by dbisu